rust-embedded/heapless #509: use of undeclared crate or module `heapless`

Admsher, 2024-08-21:
I am getting an issue in which i am not able to import heapless in my project,  have done the following:

used cargo add heapless to add the dependency (v0.8.0)

imported the crate using "use  heapless::Vec;"

also i have added #[no_std] in the file

newAM, 2024-08-21:
Please include your Cargo.toml (or ideally a commit hash to the complete code on GitHub) and the full error message.

Admsher, 2024-08-22:
Hi thanks for the reply as I don't have access to my work laptop as of now I will do it by tomorrow

Admsher, 2024-08-23:
Cargo.toml
`[package]
name = "EMDU"
version = "0.1.0"
edition = "2021"

[dependencies]
cortex-m-semihosting = "0.5.0"
embedded-storage = "0.3.1"
heapless = "0.8.0"
panic-halt = "0.2"
cortex-m-rt = "0.7"

[dependencies.MXP1064]
version = "0.1.0"
path= "../MXP1064"
features=["rt"]

[build-dependencies]
bindgen = "0.65.1"
cc = "1.0"

[features]
default = []
flexible_data_rate = []
no_superv_support = []
doze_mode_support = []
pn_mode = []
memory_error_control = []
enhanced_bit_timing_reg = []`

error

` use heapless::String;
   |     ^^^^^^^^ use of undeclared crate or module heapless`

Admsher, 2024-08-23:
Solved it , apparently we have to add the heapless in [build-dependencies] instead of [dependencies] which is strange because the cargo add imports it as a dependency

closed 2024-08-23 by Admsher