rust-embedded/heapless #510: Debug build error: no_std + FnvIndexMap (linker err: `rust_eh_personality`)

cjrh, 2024-09-11:
TL,DR, A debug build using `FNVIndexMap` fails linking with missing `rust_eh_personality`, but release build works.

Apologies if this is user error, I am new to no_std.

# Environment

<details>
<summary>See details</summary>

```
$ rustc --version --verbose
rustc 1.81.0 (eeb90cda1 2024-09-04)
binary: rustc
commit-hash: eeb90cda1969383f56a2637cbd3037bdf598841c
commit-date: 2024-09-04
host: x86_64-unknown-linux-gnu
release: 1.81.0
LLVM version: 18.1.7

$ cargo --version --verbose
cargo 1.81.0 (2dbb1af80 2024-08-20)
release: 1.81.0
commit-hash: 2dbb1af80a2914475ba76827a312e29cedfa6b2f
commit-date: 2024-08-20
host: x86_64-unknown-linux-gnu
libgit2: 1.8.1 (sys:0.19.0 vendored)
libcurl: 8.8.0-DEV (sys:0.4.73+curl-8.8.0 vendored ssl:OpenSSL/1.1.1w)
ssl: OpenSSL 1.1.1w  11 Sep 2023
os: Linux Mint 22.0.0 (wilma) [64-bit]

$ rustup show
Default host: x86_64-unknown-linux-gnu
rustup home:  /home/caleb/.rustup

installed toolchains
--------------------

stable-x86_64-unknown-linux-gnu (default)
nightly-2024-06-08-x86_64-unknown-linux-gnu
nightly-x86_64-unknown-linux-gnu
1.73.0-x86_64-unknown-linux-gnu
1.80.1-x86_64-unknown-linux-gnu
1.80-x86_64-unknown-linux-gnu

installed targets for active toolchain
--------------------------------------

x86_64-unknown-linux-gnu
x86_64-unknown-linux-musl

active toolchain
----------------

stable-x86_64-unknown-linux-gnu (default)
rustc 1.81.0 (eeb90cda1 2024-09-04)

$ uname -a
Linux caleb-ThinkPad-X13-Gen-1 6.8.0-44-generic #44-Ubuntu SMP PREEMPT_DYNAMIC Tue Aug 13 13:35:26 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux
```

</details>

# Reproducer

Here is a tiny project:

Cargo.toml:

```toml
[package]
[... 120 lines cut ...]

    // let mut h = heapless::FnvIndexMap::<i32, i32, 16>::new();
    // h.insert(1, 2).unwrap();
    0
}
```

```
$ cargo build
   Compiling heapless_debug_err v0.1.0 (/home/caleb/tmp/no_std_heapless/libc_print_how)
warning: unused import: `libc`
 --> src/main.rs:5:5
  |
5 | use libc;
  |     ^^^^
  |
  = note: `#[warn(unused_imports)]` on by default

warning: `heapless_debug_err` (bin "heapless_debug_err") generated 1 warning (run `cargo fix --bin "heapless_debug_err"` to apply 1 suggestion)
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 0.17s
```

The complaint about `rust_eh_personality` is something I've seen before, in a different library that supports no_std. In that case, it turned out that a macro using `format_args!` was somehow causing a `debug_assert` to get compiled in.  Even though my crate specifes `panic = "abort"`, nevertheless somehow the debug_assert was bringing in panic unwinding code. For _that_ library, I could resolve the problem with `debug-assertions = false`, which is why I have that set here. However, with the heapless hashmap shown here it seems to be a different issue.

So my guess is that there is something in `heapless::FnvIndexMap` that is somehow including panic code but only in debug mode.

rjsberry, 2024-09-15:
The issue here is that the stock libcore is built with panic=unwind.

If you want to stay on stable you could try bumping the opt level:

```
[profile.dev]
opt-level = 1
panic = "abort"
debug-assertions = false

[profile.release]
panic = "abort"
```

If you can use a nightly toolchain compile libcore yourself with panic=abort:

```
cargo +nightly -Zbuild-std build --target x86_64-unknown-linux-gnu
```

Related: https://github.com/rust-lang/rust/issues/56152

Dirbaio, 2024-09-15:
closing, not a bug in heapless as per the above comment.

closed 2024-09-15 by Dirbaio