rust-embedded/heapless #518: no_std use core::error::Error 1.81 stabilizes the Error trait in core

zhang-wenchao, 2024-11-01:
1.81 stabilizes the Error trait in core, allowing usage of the trait in #![no_std] libraries.

zhang-wenchao, 2024-11-01:
https://github.com/dtolnay/anyhow/issues/389

Dirbaio, 2024-11-01:
can you expand on what types from this lib would you expect to implement `core::error::Error`? This is not very actionable otherwise.

zhang-wenchao, 2024-11-01:
I just want to simply use `?` operator to propagate errors. Don't map_err.

```
use heapless::String;
use anyhow::Result;

pub fn test() -> Result<String<256>> {
    let mut phrase = String::new();
    phrase.push_str("XX")?;
    Ok(phrase)
}
```

```
the trait bound `(): core::error::Error` is not satisfied
the trait `FromResidual<Result<Infallible, E>>` is implemented for `Result<T, F>`
required for `anyhow::Error` to implement `From<()>`
required for `Result<String<256>, anyhow::Error>` to implement `FromResidual<Result<Infallible, ()>>`rustc[Click for full compiler diagnostic](rust-analyzer-diagnostics-view:/diagnostic message [0]?0#file:///home/life/Develop/job/my/embedded/ohw-wallets/src/mnemonic.rs)
```

```
pub fn test() -> Result<String<256>> {
    let mut phrase = String::new();
    phrase.push_str("XX").map_err(|_| anyhow!(""))?;
    Ok(phrase)
}
```

closed 2025-03-28 by newAM