rust-embedded/heapless #519: Implement bincode Writer and Reader traits for Vec

petekubiak, 2024-11-29:
I am working on a no_std embedded comms stack which talks to devices running C applications. The stack uses `heapless::Vec`s to handle the byte streams being sent and received, and I would like to use `bincode` to perform the conversion to / from Rust structures.

Right now, in order to do this I have to create a wrapper type around the `heapless::Vec` in order to implement `bincode`'s `Writer` and `Reader` traits. It seems like this is a common enough pairing that it would make sense to have these traits implemented for `heapless::Vec` by default, or at least available through a feature, rather than having to re-implement for each new project working in this space.

I'm not sure whether it would make more sense for this implementation to live in the `bincode` crate or the `heapless` crate, so I've created [an equivalent issue on the `bincode` repo](https://github.com/bincode-org/bincode/issues/733). What do you think?

petekubiak, 2024-11-29:
Yeah agreed, to me it would seem like it would make more sense for the implementation to live in the serialiser's crate rather than the collection's crate. Good to know there's precedent.

petekubiak, 2025-07-04:
Well if the bin code maintainers are not interested in adding heapless support I guess it's up to the heapless maintainers whether they want to add bincode support...that does seem the wrong way round to me though.

David-OConnor, 2025-07-18:
It seems we are at an impasse. I can write a third party integration crate if we have to, but would prefer to resolve this through diplomacy! (C+P from there)