rust-embedded/heapless #529: 0.9.0 Release

NWPlayer123, 2025-02-25:
What is there still to be done to release a new version of heapless?

There's a pretty sizeable [changelog](https://github.com/rust-embedded/heapless/blob/main/CHANGELOG.md#unreleased), and I've had to pin my heapless dependency on a project to the latest git release since I have a need for Vec::drain.

I also encountered compile errors trying to construct a String::from_str due to String now being a type alias for a StringInner compared to crates that are still targeting the 0.8.0 release with an incompatible type (e.g. embedded_svc::wifi::ClientConfiguration).

sosthene-nitrokey, 2025-02-25:
- https://github.com/rust-embedded/heapless/pull/502

Should be merged before a 0.9 release to reduce the risk of breakage of existing code.

> I also encountered compile errors trying to construct a String::from_str due to String now being a type alias for a StringInner compared to crates that are still targeting the 0.8.0 release with an incompatible type (e.g. embedded_svc::wifi::ClientConfiguration).

I introduced the change that made `String` an alias for `StringInner` , but I don't understand that breakage. Could you please give an example?

NWPlayer123, 2025-02-25:
> I introduced the change that made `String` an alias for `StringInner` , but I don't understand that breakage. Could you please give an example?

I double checked the error; from_str no longer exists so I changed it to `ClientConfiguration { ssid: heapless::String::<32>::from_utf8("MyNetwork").expect("Error converting WIFI_SSID") }` which tells me mismatched types: expected `heapless::string::String<32>`, found `StringInner<OwnedStorage<32>>` due to the type no longer existing between 0.8 and main, so until 0.9 is released and crates update the dependency, I have to do `"MyNetwork".try_into().expect("Error converting WIFI_SSID")`. Not a huge problem, just annoying to have to do

sosthene-nitrokey, 2025-02-25:
Are you directly depending on the updated heapless with

```toml
[dependencies]
heapless = { git = "https://github.com/rust-embedded/heapless.git" }
```

?

That leads to duplicate dependencies, and you probably would still have the same error even without the `StringInner` refactor.

Instead of depending on heapless through git, you maybe can patch it:

```toml
[dependencies]
heapless = "0.8.0"

[patch.crates-io]
heapless = { git = "https://github.com/rust-embedded/heapless.git" }
```

zeenix, 2025-03-03:
> * [Re-introduce covariance for containers #502](https://github.com/rust-embedded/heapless/pull/502)
> 
> 
> Should be merged before a 0.9 release to reduce the risk of breakage of existing code.

@sosthene-nitrokey That's fair. I don't intent to put any pressure on you (or anyone) but if that looks like it will take some time, would you consider creating a 0.8.0 branch with some fixes (I'm thinking of #530) applied and/or backported and doing a 0.8.1 first?

sosthene-nitrokey, 2025-03-03:
I'm not the maintainer of this library, I can't prepare releases. At Nitrokey we'd like to see #502 be merged and released when possible too. From my point of view the PR is ready for merging.

My understanding is that a 0.8.1 release could include #502 and the `View` types and that it does not include breaking changes. No need to backport anything.

newAM, 2025-03-09:
We have also been discussing a 1.0 release, but that's a bit further in the future to get dependencies to 1.0 and everything in #494.

I'm not opposed to a 0.9 until then, what do you think @rust-embedded/libs ?

closed 2025-04-28 by reitermarkus

zeenix, 2025-06-15:
The 0.9.0 release turned out to be broken and had to be yanked, so we're still stuck with the old 0.8. :( Please re-open this until 0.9.1 can be released.

PR #556 (2025-04-28): Release 0.9.0
Fixes #529.
commit 92b476a4 (2025-04-28): Release 0.9.0

Fixes #529.