rust-embedded/heapless #535: Heapless 1.0

newAM, 2025-03-09:
General discussion / notes / thoughts about stabilizing heapless with a 1.0 release.

---

# Checklist

- [ ] Review issues in the [v1.0.0 milestone](https://github.com/rust-embedded/heapless/milestone/3)
  - Planned breaking changes are tracked in #494 
- [ ] Review [Rust API Guidelines Checklist](https://rust-lang.github.io/api-guidelines/checklist.html#rust-api-guidelines-checklist)
- [ ] Review allowed warnings and clippy lints
- [ ] When 1.0 looks ready engage with community, solicit feedback
  - [ ] Announce in rust-embedded WG meeting
  - [ ] Announce in this issue
  - [ ] Anywhere else?

---

# Dependencies

- defmt will reach 1.0 soon: https://ferrous-systems.com/blog/defmt-1-0/
- ufmt may not be 1.0, leave as-is and mention it will not be covered by semver?
- hash32 needs a 1.0

rursprung, 2025-03-20:
see also the discussion on #410 - i think you wanted to push `hash32` to 1.0 first (https://github.com/rust-embedded-community/hash32/issues/17)?

newAM, 2025-03-24:
Whoops I forgot #410 existed, sorry about that!

reitermarkus, 2025-04-03:
Got two more issues:

- [ ] Remove `Q*` type aliases for `MpMcQueue`, and rename it to just `Queue` to align with `spsc::Queue`.
- [ ] Don't export `indexmap` types at the top-level, except `indexmap::IndexMap` itself. Also, should the module be called `index_map` instead?

pvdrz, 2025-04-04:
> Got some more issues:
> 
>     * [ ]  Remove `Q*` type aliases for `MpMcQueue`, and rename it to just `Queue` to align with `spsc::Queue`.[ ]  Don't export `indexmap` types at the top-level, except `indexmap::IndexMap` itself. Also, should the module be called `index_map` instead? Same for `indexset` module and `IndexSet` types.[ ]  Align `histbuf` module and `HistoryBuffer` type name as `history_buf` and `HistoryBuf`?

See https://github.com/rust-embedded/heapless/pull/547

Conaclos, 2026-01-21:
> Can you tell me what the impact is for Debian?

Not so important, because the last Debian stable supports rustup. However, this is still a way of choosing a MSRV.