rust-embedded/heapless #554: Possible data race in the iteration of Stack's pop and UnionNode

thalesfragoso, 2025-04-13:


jr-oss, 2025-04-14:
AFAIUI `top_addr` will stay the same, even when some other task adds or removes an element.
`top_addr` holds the address of a pointer that points to the first Node, i.e. updates to `*top_addr` must be atomic, which is achieved by `load_link`and `store_conditional`.
Disclaimer: I am not an expert on atomic operations and might be totally wrong

reitermarkus, 2025-04-26:
Took me a while to understand the problem.

`llsc.rs`

```rust
92            let top = arch::load_link(top_addr);
              // `try_pop()` may be called in other thread.
94            if let Some(top) = NonNull::new(top as *mut N) {
                  // `try_pop()` may be called in other thread.
95                let next = &top.as_ref().next();
```

`cas.rs`

```rust
216        if let Some(mut top) = stack.top.load(Ordering::Acquire) {
               // `try_pop()` may be called in other thread.
217            let next = unsafe { top.non_null().as_ref().next().load(Ordering::Relaxed) };
```

So calling `as_ref()` is still sound, given that the `top` node lives as long as the stack itself.

Calling `next()` is unsound for `UnionNode` since another thread may already have written the `data` variant.

thalesfragoso, 2025-04-27:
I think even `as_ref` for `UnionNode` is unsound under stack borrows.

I was working into getting some loom + miri support for heapless. My idea is to have a PR with a failing loom test and then provide a fix, that way we can hopefully avoid such unsoundness in the future too.