rust-embedded/heapless #562: Document all `unsafe` blocks.

reitermarkus, 2025-04-26:
All `unsafe` blocks should be documented and be scoped as closely as possible.

After all are documented, the `clippy::undocumented_unsafe_blocks` lint should be enabled.

Mortano, 2025-08-13:
I should be able to help with that, if the help is still wanted. Are there any contribution guidelines for first-time contributors?

Mortano, 2025-08-13:
Ok I'll start working on this then :)

zeenix, 2025-08-16:
> Do you have a preference as to how detailed the documentation should be?

I can't speak for @reitermarkus here but IMO while your attention to detail is appreciated, I really don't think we need this level of detail. Typically, you'd just have a bullet list for each safety requirement mentioned in the documentation of the API being used, shortly explaining why/how the requirement is satisfied.

I [typically keep them short](https://github.com/zeenix/zlink/blob/2e33c64e2d770397737f45dd6931dba439331b3d/zlink-core/src/server/mod.rs#L94) but probably that's a bit too short. :)

Mortano, 2025-08-17:
Ok I'll try to keep it more concise then, thanks for the clarification

Mortano, 2025-08-20:
I found one potential safety violation related to #579 (see my comment in there). I'll continue checking and commenting the rest of the `unsafe` code and will comment here if I find something else, then we can discuss how to deal with each issue.