rust-embedded/heapless #568: `LenType` causes regression regarding supported `Vec` sizes and usage in libraries.

sosthene-nitrokey, 2025-04-29:
The addition of `LenType` to vecs and the `DefaultLenType` means that arbitrary length of `Vec` are much less usable.

# Not all lengths "just work" with `Vec` like they did previously

For example, the  following: `Vec<u8, 587>` does not compile anymore, and needs to be worked around with `Vec<u8, u16, 587>`.

This itself is I think a significant downgrade in readability (the benefits of `LenType` are in most cases I think not worth making code less readable, since the overhead of the Vec itself will often be negligible compared to the buffer of the Vec).

# In a library setting it forces the library writer to be explicit of the `LenType`:

```rust
fn some_func<const N: usize>() -> Vec<u8, N> {
    todo!()
}
```

Fails to compile with

```
```

sosthene-nitrokey, 2025-04-29:
And actually I thought the next release would have been 0.8.1, not a breaking release. In my PR I made efforts to make sure that there were no breaking changes, and documented the breaking changes that were pending, but they were not implemented once it was decided to go with a 0.9.0 instead of 0.8.1 release: https://github.com/rust-embedded/heapless/issues/494

I saw the PR for the 0.9 release I should have remembered this.

newAM, 2025-04-29:
This is a fairly serious issue that is going to break a lot of code.

I have yanked 0.9.0 from crates.io for now to save people the effort of attempting to update their code.

reitermarkus, 2025-04-29:
I wouldn't really call this a regression since `LenType` is working as designed, and is a breaking change and a bit less ergonomic, as was to be expected.

> `Vec<u8, 587>` does not compile anymore, and needs to be worked around with `Vec<u8, u16, 587>`.

We can add more sizes, e.g. all up to 1024 to avoid needing the explicit type in most common cases.

> In a library setting it forces the library writer to be explicit of the `LenType`:

Is that a bad thing? A library author should know whether `usize` is needed or e.g. `u8` is sufficient for a particular function.

> It makes compatibility with `View` types much harder.

Not that much harder:

```rust
let vec: &mut VecView<_, _> = &mut Vec::<u8, 8>::new();
```

Note this isn't an argument for or against anything, just a clarification that it isn't as dramatically broken as this issue suggests.

sosthene-nitrokey, 2025-04-29:
I don't think it's "broken". But it will cause a lot of code churn, for a feature I don't think many users need. At least we don't, we much more concerned about compile times, binary size, and reducing the use of generics in our application code than the small memory overhead this might save (if it even saves anything given the).

I suggested yanking because I don't think it was understood that the disruption would be this great.
Without the `DefaultLenType` mechanism, most crates could have upgraded by just bumping the number. Now it means almost every downstream will have to adapt their code, adding generics, that add noise to the code and increase monomorphization.

tomkris, 2025-05-05:
There is a way to auto implement this kind of len for any capacity using unstable `generic_const_exprs`

https://play.rust-lang.org/?version=nightly&mode=debug&edition=2024&gist=1fd5196d7b7774192aab4aea17b13a11

Unfortunately I cannot make compiler to believe that implementation exists for all possible `C: usize`, so constraint on capacity has to be added on every struct where `Vec<T, C>` is used without explicitly specifying `C` as constant; this makes use of this approach even harder.

If we cannot figure out how to make len auto-size on all possible inputs, I thin kit should default to `usize` to keep backwards compatibility with existing code and maintain ergonomics.

Personally I'm very interested to have auto-size len though. I have highly memory constrained workload, some of my dependencies use `heapless::Vec` extensively.

sosthene-nitrokey, 2025-05-28:
Closed by https://github.com/rust-embedded/heapless/pull/569