rust-embedded/heapless #598: Performance regression for `HistoryBuf(fer)::write()` with heapless 0.9.1

BVollmerhaus, 2025-08-24:
## Description

I've been using heapless's `HistoryBuffer` (now renamed to `HistoryBuf`) as a sliding search window with much success. More precisely, I have an iterator over a substantial number of bytes and need to find a particular sequence, so I'm doing something akin to the following:

```rust
let mut search_window: HistoryBuf<u8, 11> = HistoryBuf::new();

for byte in lots_of_bytes {
    search_window.write(byte);

    if search_window.oldest_ordered().eq(b"some-needle") {
        // Needle found; do something useful
        return;
    }
}
```

While I'm not using heapless in its intended embedded use case, I found its `HistoryBuffer` to be much faster than using a `VecDeque` with a manual `pop_front()`/`push_back()` or similar alternatives.

However, with heapless 0.9.1, there seems to be a significant regression in the performance of the above snippet, which I'd largely attribute to the `write()` call. Of course, the `oldest_ordered().eq()` in every iteration has a proportionately larger impact on runtime than the `write()` on its own, but simply upgrading heapless to 0.9.1 with this snippet still results in about 400 MB/s less throughput in a real-world example (~1.4 → ~1.0 GB/s).

Different optimization settings like LTO or codegen units do not seem to impact the result much. You can find a Divan-based benchmark as an MRE below. This may very well be a case of me holding it wrong, but I figured it might be worth reporting nonetheless. 🙂

## Environment

* **CPU:** Ryzen 7 PRO 5850U
* **Kernel Version:** 6.16.1
* `rustc -vV`

   ```text
   rustc 1.89.0 (29483883e 2025-08-04)
   binary: rustc
   commit-hash: 29483883eed69d5fb4db01964cdf2af4d86e9cb2
   commit-date: 2025-08-04
   host: x86_64-unknown-linux-gnu
   release: 1.89.0
   LLVM version: 20.1.7
   ```

## MRE Benchmark

```rust
// benches/historybuf.rs

use std::hint::black_box;

use divan::counter::BytesCount;
use heapless::HistoryBuf;  // Replace with `HistoryBuffer` for heapless 0.8.0

const NEEDLE: &[u8] = b"needle451";
const WINDOW_SIZE: usize = NEEDLE.len();

fn historybuf_search_window_write(data: Vec<u8>) {
    let mut search_window: HistoryBuf<u8, WINDOW_SIZE> = HistoryBuf::new();

    for byte in data {
        search_window.write(byte);
    }
}

#[divan::bench]
fn bench_historybuf_search_window_write(bencher: divan::Bencher) {
    let total_bytes: usize = 8 * 1024 * 1024;
    let bytes_before_needle = total_bytes - NEEDLE.len();

    bencher
        .counter(BytesCount::new(total_bytes))
        .with_inputs(|| {
            let mut data = Vec::with_capacity(total_bytes);
            data.extend(vec![b'\0'; bytes_before_needle]);
            data.extend(NEEDLE);
            data
        })
        .bench_values(|data| {
            historybuf_search_window_write(black_box(data));
        });
}

fn main() {
    divan::main();
}

```

```toml
# Cargo.toml

[package]
name = "heapless-perf-mre"
edition = "2024"

[dependencies]
heapless = "0.8.0"

[dev-dependencies]
divan = "0.1.21"

[profile.release]
lto = "fat"
codegen-units = 1

[[bench]]
name = "historybuf"
harness = false
```

### Results

#### heapless 0.8.0 (`HistoryBuffer`)

`cargo bench --benches`

```text
     Running benches/historybuf.rs (target/release/deps/historybuf-2a85391490864b67)
Timer precision: 20 ns
historybuf                               fastest       │ slowest       │ median        │ mean          │ samples │ iters
╰─ bench_historybuf_search_window_write  24.61 µs      │ 275.3 µs      │ 119.5 µs      │ 124.4 µs      │ 100     │ 100
                                         340.7 GB/s    │ 30.46 GB/s    │ 70.15 GB/s    │ 67.37 GB/s    │         │
```

#### heapless 0.9.1 (`HistoryBuf`)

```text
     Running benches/historybuf.rs (target/release/deps/historybuf-974f73ca8d39caa1)
Timer precision: 20 ns
historybuf                               fastest       │ slowest       │ median        │ mean          │ samples │ iters
╰─ bench_historybuf_search_window_write  6.823 ms      │ 7.078 ms      │ 6.87 ms       │ 6.892 ms      │ 100     │ 100
                                         1.229 GB/s    │ 1.185 GB/s    │ 1.22 GB/s     │ 1.217 GB/s    │         │
```

zeenix, 2025-08-24:
> However, with heapless 0.9.1, there seems to be a significant regression

Wow, the regression indeed is a quite severe. 😢 If we had benchmarks in the repo and ran it as part of the CI (like I do in [zbus project](https://github.com/dbus2/zbus/blob/main/.github/workflows/bench.yml)), it would help identifying such regressions early on. Do you think you could contribute that (I'm asking cause it seems you're already half way there since you've some benchmarks)?

As for the actual issue, I've never used this part of the API but I guess someone will need to do a `git bisect` with the benchmarking code you provided to find out which change introduced this and hopefully the person who was responsible, will be interested in fixing this regression too. :)

jannic, 2026-10-27:
To make it easier to get a disassembly of the relevant code, I added `#[inline(never)]` on `historybuf_search_window_write`.
With that, and heapless 0.8.0, I got this nice and short result:

```
.section .text.historybuf::historybuf_search_window_write,"ax",@progbits
	.prefalign	4, .Lfunc_end4, nop
	.type	historybuf::historybuf_search_window_write,@function
historybuf::historybuf_search_window_write:
	.cfi_startproc
	cmp qword ptr [rdi], 0
	je .LBB4_1
	mov rdi, qword ptr [rdi + 8]
	jmp qword ptr [rip + free@GOTPCREL]
.LBB4_1:
	ret
```

With 0.9.1, I got this long assembly instead:
```
.section .text.historybuf::historybuf_search_window_write,"ax",@progbits
	.prefalign	4, .Lfunc_end4, nop
	.type	historybuf::historybuf_search_window_write,@function
historybuf::historybuf_search_window_write:
	.cfi_startproc
	.cfi_personality 155, DW.ref.rust_eh_personality
	.cfi_lsda 27, .Lexception1
	push r15
	.cfi_def_cfa_offset 16
	push r14
	.cfi_def_cfa_offset 24
	push rbx
	.cfi_def_cfa_offset 32
	.cfi_offset rbx, -32
	.cfi_offset r14, -24
	.cfi_offset r15, -16
	mov rax, rdi
	mov r15, qword ptr [rdi]
	mov rdi, qword ptr [rdi + 8]
	mov rcx, qword ptr [rax + 16]
	test rcx, rcx
	je .LBB4_4
	xor esi, esi
	xor eax, eax
	.p2align	4
.LBB4_2:
	cmp rax, 8
	ja .LBB4_9
	mov rdx, rax
	inc rdx
	cmp rdx, 9
	sete r8b
	mov eax, 0
	cmovne rax, rdx
	or sil, r8b
	dec rcx
	jne .LBB4_2
.LBB4_4:
	test r15, r15
	je .LBB4_11
	pop rbx
	.cfi_def_cfa_offset 24
	pop r14
	.cfi_def_cfa_offset 16
	pop r15
	.cfi_def_cfa_offset 8
	jmp qword ptr [rip + free@GOTPCREL]
.LBB4_11:
	.cfi_def_cfa_offset 32
	pop rbx
	.cfi_def_cfa_offset 24
	pop r14
	.cfi_def_cfa_offset 16
	pop r15
	.cfi_def_cfa_offset 8
	ret
.LBB4_9:
	.cfi_def_cfa_offset 32
	mov rbx, rdi
	lea rcx, [rip + .Lanon.56581d111967526f5d973bbc1dbd6320.12]
	lea rdx, [rip + .Lanon.56581d111967526f5d973bbc1dbd6320.13]
	test sil, 1
	cmovne rdx, rcx
	mov esi, 9
	mov rdi, rax
	call core::panicking::panic_bounds_check
	ud2
	mov r14, rax
	test r15, r15
	je .LBB4_8
	mov rdi, rbx
	call qword ptr [rip + free@GOTPCREL]
.LBB4_8:
	mov rdi, r14
	call _Unwind_Resume@PLT
```

(Both with rustc 0.99.1 on x86_64.)

jannic, 2026-10-27:
> 10\. With that, and heapless 0.8.0, I got this nice and short result:

With the heapless version from #691 (commit 761d12f), I get exactly the same assembly as with 0.8.0.